astrohr/dagor_tca #16: Mega (dome MCU) goes nuts when server isn't on

frnhr, 2016-05-28:
Seems like USB extenders start to draw too much power. Unplugging USB resolves the problem.

 - [ ] Test possible voltage differences
 - [ ] Check how US extenders behave in different power situations
 - [ ] Would a hub help?

vedran-dobos, 2016-06-20:
Test again after fixing issue #24 - Arduino Mega (C1) was not properly (commonly) grounded.

frnhr, 2016-07-19:
Still the same.

Looks like server draws power via USB when not powered on.

Putting a USB opto isolator between server and C1 (dome Arduino controller) fixes the problem, but brings up another one: Server doesn't recognize USB devices behind Opto_Isolator + Hub combo. It works fine with only one device, but not with both. My laptop recognizes all devices regardless. I believe a powered USB hub would solve the issue. The optoisolator *is* powered, but it is probably not very high quality.

For reference, check the [Overview diagram](https://github.com/astrohr/dagor_tca/wiki#overview).

[image]

frnhr, 2016-07-19:
Also, the USB hub seems to be damaged, only two of four ports work (when tested directly with a laptop). I have replaced it with another USB hub, also not powered.

frnhr, 2016-07-21:
Fixed by adding an opto isolator:

[image]

... and a powered USB hub:

[image]

closed 2016-07-21 by frnhr