astrohr/dagor_tca #17: Mega (dome MCU) triggers dome rotation randomly

frnhr, 2016-05-28:
Possibly related to some ground issues, or possibly a blown internal pullup in Arduino.

 - [ ] Plugin dongles in ports on dome enclosure
 - [ ] Try without any dongles plugged in
 - [ ] Plug one dongle at a time
 - [ ] Check voltage levels in manual control boxes (those two yellow dongles with 4 buttons)
 - [ ] Add an external pullup resistor on Mega inputs
 - [ ] ... or change input pins (just for the test)

frnhr, 2016-05-28:
##### Plugin dongles in ports on dome enclosure
No effect.

##### Try without any dongles plugged in
No random rotations, unless Mega is touched with a finger. So almost no effect...

vedran-dobos, 2016-06-20:
Test again after fixing issue #24 - Arduino Mega (C1) was not properly (commonly) grounded.

frnhr, 2016-08-05:
It was the ground...

closed 2016-08-05 by frnhr