astrohr/dagor_tca #19: Doors Nano (dome doors MCU) no longer loses power when using manual switch

frnhr, 2016-05-28:
Is this a bug or a feature? 😄

vedran-dobos, 2016-05-28:
That's a (new) feature! :smile: 

The relays still do lose power when the manual switch is not in the "0" position, so it should be impossible to short-circuit anything.

If we add new wires to the analog input of the Dome doors MCU through a voltage divider this might allow us to:
- know if the manual switch is not in the "0" position, and report it to the user/software
- measure the voltage and thus estimate the battery charge/state, and warn the user/software that charging of the battery might be needed, or there is another problem with the battery 
(see: http://www.instructables.com/id/12V-Battery-State-Of-Charge-Indicator/)

frnhr, 2016-05-28:
Perfect!

Re battery meter, that's a good idea, thought we have more pressing concerns at the moment. Created a separate issue for it: https://github.com/astrohr/dagor_tca/issues/28

closed 2016-05-28 by frnhr