astrohr/dagor_tca #24: Power regulators in dome encl. are too hot

frnhr, 2016-05-28:
20-10-5-0 V regulators are too hot.

 - [ ] What draws too much current
 - [ ] Do we need a better circuit?

vedran-dobos, 2016-06-14:
Just for testing purposes the 5V rail is bypassed, and it actually fixed the voltage problems.

[image]
[image]

The 20V, 10V, and 5V rails are now within a few % error.
[image]
[image]
(20V rail not shown, but is also OK)

There might be problems within the cable leading to the dome encoder and position switches.
I will rework all the connections and identify the culprit (encoder/switches/something_else).

vedran-dobos, 2016-06-20:
Found the issue!! :satisfied:
Arduino Mega (C1) was not properly (commonly) grounded.

Added a ground wire, and most of the problems went away. It could be done more nicely, but works anyway.

[image]

Output is now stable:
5V:
[image]

10V:
[image]
20V:
[image]

Still, the analog inputs are not working properly, too small resistance measured, resulting in increased power consumption:
A4 = 175 Ohm
A5 = 142 Ohm
A6 = 290 KOhm
All other analog inputs ~7 MOhm

- [ ] Use different analog inputs or switch to digital inputs (if possible)

Other fixes done while debugging:

- Changed C1 power adapter setting to 4.5V, as in reality it is delivering 7.7V at this setting (under load). Setting on 7.5V delivered above 11V, and caused unnecessary heating of the LM7805 on the C1.
[image]

- Simplified connection to dome sensor cable
[image]

- Split sensor cable for dome rotation encoder/fixed switches in an electric box
[image]
[image]
Note: pinout changed after photo taken.

- Redone connection to encoder with nice heat-shrink tubing :heart_eyes:
[image]

frnhr, 2016-06-20:
Nice work!

Re common ground - makes perfect sense! Mega ("C1") used to be powered by the common 5V output (and grounded there too). For _reasons_ (keeping the heat down, I guess) we added the extra switching power source, but no-one thought of grounding its output :)

Re analog inputs - that would imply faulty (i.e. burned) Atmega chip, since normally impedance on all inputs is very high, of course. BTW datasheet actually says it should be 100MΩ, but that's probably not realistic, as discussed here: http://electronics.stackexchange.com/a/67173/19509

We can't switch to digital inputs without loosing the two-step speed control (well, we _could_, with a few voltage comparators... but let's keep it simple). We have another Arduino Mega board, so we can test how it behaves. And we can switch to using the other analogue inputs. And buy a new Mega...

vedran-dobos, 2016-06-20:
Re common ground: Took a whole day of figuring out how to connect one simple wire in the end :sweat_smile:

Re analog inputs: The A4-6 pins are used for the dome rotation switches, not for the speed control...?
Anyhow, let's change the A inputs and change the code to see how it behaves.

- [ ] Test position switches and dome encoder through software to ensure cabling was re-done correctly

frnhr, 2016-06-22:
Ah, those ADC inputs, I see.

Actually, adding 3 x 5kΩ in front of those pins would most likely solve our problem.
At the same time, the Mega shield needs to be made as a PCB (it is currently made on a proto-shield). I'll make a new issue for that, so feel free to close this one if there is nothing else to be done.

closed 2017-03-28 by frnhr