astrohr/dagor_tca #26: Doors Nano (MCU for controlling dome doors) needs signals from limit switches

frnhr, 2016-05-28:
Max voltage for Arduino: 5V!

Range for a logical "1": 3V - 5V
Range for logical "0": 0V - 1.5V
(a simple"voltage divider" in 20k - 100k range should be ok)

 - [ ] What is the impedance of digital input or ATMega328?

vedran-dobos, 2016-05-28:
Cabling done. Ends in the control box.
It's an independent Normally Open circuit until the switches engage, at that point the circuit closes.

Used the 2 pairs of signal wires:
-blue/white-blue for "open" switch
-green/white-green for "closed" switch

vedran-dobos, 2016-05-31:
Connected MCU C3 with doors sensors.
Pin D4 - Doors closed
Pin D3 - Doors opened

Code here on Github shiws using analog inputs?
Is this old code, or some other error?

closed 2016-06-03 by frnhr