astrohr/dagor_tca #31: Cooling enclosure in control room

vedran-dobos, 2016-05-28:
Better cooling of the 3-phase motor controllers.

Big hole done on top, and better/stronger fan with protection grill mounted. But fan is not working at the moment.

If possible, use controller (Arduino?) to change fan speed (and/or start-stop) depending on temperature inside.

There is a thermostat/fan module in a shipping container near the dorm. Use that in the beginning maybe?

frnhr, 2016-05-29:
Mounted two fans, and a dist filter to prevent stuff falling in because stepper drivers are directly below the vent. The two fans are connected in series on 12V. Noise level is fairly low this way (about 45 db near the PC monitor). Connecting them to 15V increases noise level to 50db. We are probably ok with 12V though, with respect to cooling. Max temperature inside the enclosure was 27ºC, after several minutes of full-speed teleskope movements.

[image]

Also cleared metal debris inside the drivers. Nothing much, but there were some very fine chips in, almost dust. 
If you always wanted to know what is inside those driver, but were afraid to ask:
[image]
[image]

frnhr, 2016-05-29:
Todo:

- [ ] better PCB for voltage regulators inside the breaker box (currently it's a breadboard)

frnhr, 2016-05-30:
Proto-PCB is ready:
[image]

Nothing special here, a voltage regulator and two connectors for fans, in series. Two "features:
 * Jumpers can be set so that fans are connected in parallel or in series (photo shows config for series)
 * Voltage regulator is plugged in a 3-pin header, and is removeable. Any 78xx will do. Input is 24V.

Drawing a schematics is probably an overkill. But a nice label would be neat...

 - [ ] Mount it inside the breaker box

closed 2016-06-03 by frnhr