astrohr/dagor_tca #6: calculating/interpolating dome position to match with telescope

frnhr, 2016-05-28:


vedran-dobos, 2016-05-30:
Running a neural network on the collected data did not yield satisfactory results.
More manual input (learning) data might be needed, or a different approach altogether.

Or we can try to mathematically calculate the position through a mysterious formula :smile: 

[image]

[altazazch.txt](https://github.com/astrohr/dagor_tca/files/289131/altazazch.txt)

frnhr, 2016-05-30:
Best way to plot this data is in cylindrical coordinates, with telescope position on polar plane (azimuth on θ, 90°-altitude on r) and dome azimuth on z axes.

Probably won't make a difference in the NN result, just putting it here for future reference.

closed 2017-03-28 by frnhr