astrohr/dagor_tca #37: Fans for primary mirror

frnhr, 2016-05-29:
4 fans should be placed on the base ________, on plastic caps behind primary mirror.

 - [ ] mount
 - [ ] connect power
 - [ ] install relays and arduino control, similar to existing lights switch
 - [ ] code CLI module
 - [ ] code API module
 - [ ] code ASCOM module

vedran-dobos, 2016-05-30:
A few questions:

- what size should the fans be? could standard 12cm work?
- do fans need to be mechanically decoupled (anti-vibration mounting)?
- is speed control necessary? is it dependent on temperature difference?
- is individual speed control necessary (multi zone temp control)?
- is noise an issue?

frnhr, 2016-07-19:
Final conclusions about the fans (after multiple emails and in person talks)...

 * only 3 fans
 * blowing in
 * must have dust filters
 * we already have the fans, 12 cm, scavenged from a server rack, 6 of them
 * we have 5 dust filters
 * anti-vibration mounting is preferable
 * no need for speed reduction
 * don't care about noise

frnhr, 2016-08-11:
Temporarily mounted:
[image]

Connected to server, CLI working fine. API also tested (on the bench) and working. Next step: test ACOM.

nkatanic, 2016-08-11:
ASCOM driver tested and working.

frnhr, 2016-08-11:
Measures for caps:

Outer diameter: 205 mm
Inner diameter: 115 mm
Holes distance: 104.5 mm
Holes diagonal distance: 147.8 mm
Holes: M4 thread
Place thickness: >2mm, not important

[image]

vedran-dobos, 2016-08-11:
May I suggest taking into account rubber grommets for noise/vibration reduction into the design?

[image]

I have a few lots, and we can select the appropriate ones.

frnhr, 2016-08-11:
Can we use these and still not have gaps between fan and cap plate?

closed 2016-09-22 by frnhr