astrohr/dagor_tca #39: Labels! Moar labels!!

frnhr, 2016-05-30:
Who printed these?!
[image]

Why only these?! We need labels! Moar labels!
[image]

Can we print tiny graphics on labels?

vedran-dobos, 2016-05-30:
They must have printed themselves! :wink:

See page 14 of manual for graphics:
http://download.dymo.com/dymo/user-guides/LetraTag/LT100H/UG/LT100H_UserGuide_en-US.pdf

vedran-dobos, 2016-05-31:
These things keep reproducing by themselves!! We can't stop them anymore!!

[image]

frnhr, 2016-07-15:
### Dome fuse enclosure

 - [ ] Lower switch should be labeled "DIRECT CONTROL" instead of "ELECTRONIC CONTROL"
 - [ ] Let's change "CLOCKWISE" to "UP" and "DOWN", following the [Right-hand rule](https://en.wikipedia.org/wiki/Right-hand_rule#Rotation)

### Telescope drives enclosure

 - [ ] "MAIN POWER"
 - [ ] "DRIVES POWER"
 - [ ] "DO NOT SWITCH WHILE TELESCOPE IS MOVING!"

### Other

 - [ ] Let's label switches for dome doors with "OPEN" and "CLOSE"
 - [ ] Maybe laben enclosures themselves? Like "TELESCOPE DRIVES ENCLOSURE"...

### MOAR!!!

frnhr, 2016-09-22:
- [ ] Camera, On, Off

frnhr, 2016-09-30:
- [ ] Focuser On, Off

frnhr, 2016-09-30:
- [ ] Various labels on focuser enclosure (the black one on the spider)

closed 2018-10-07 by frnhr