astrohr/dagor_tca #47: Check what wires are used by USB extenders

frnhr, 2016-07-19:
Maybe we have available wires for power.

vedran-dobos, 2016-07-19:
Highly unlikely.

Most probably every USB wire is fed through a twisted pair on the ethernet cable, or the data+/- pair is using one twisted pair, and the other 6 wires are V+/-.

Worth checking though :smile:

frnhr, 2016-07-19:
Yeah, probably.

Ideally, though, we would have some free wires, and those would be the same wires that are used by PoE standard :)

frnhr, 2016-07-21:
Power (5 V) is connected to line `7` and `8`

Ground is connected to lines `3`, `4`, `5` and `6`.

We can make REVERSED POLARITY PoE. Not sure if we _should_... But it's an option.

Here is the USUAL PoE pinout:

[image]

There are [other standards](http://pinoutsguide.com/Net/poe_pinout.shtml), naturally...

I'm closing this issue for now. We will reopen if need PoE.

BTW, we already have a pair of PoE injectors, unused:
[image]

closed 2016-07-21 by frnhr