astrohr/dagor_tca #68: Reinstall windows on operator PC

frnhr, 2016-08-22:
Windows are slow and would probably benefit greatly from a clean install. 

There is a licence sticker on the PC, but it is an "OEM key" - this means that we cannot go to microsoft.com and download an installation DVD. Instead we should "use the one we got with the PC". Or email the vendor (IBM, I think) and ask for a new one... ... ...

I have created a "Recovery disc" ([from windows](https://support.microsoft.com/en-us/help/17423/windows-7-create-system-repair-disc)) and tried to reinstall / repair the installation this way. But:

Once I boot in the repair more, mouse and keyboard no longer work! They work in BIOS (and in boot menu), but once recovery software boots, they no longer work. I have tried switching to the 2 remaining built-in USB ports (instead of the added PCI USB card), same thing. the PC has no PS/2 ports.

Ideas welcome! 

Re:
 - license
 - keyboard & mouse problem

vedran-dobos, 2016-08-23:
Save/backup what is important, and leave the rest to me... :smile:

frnhr, 2016-08-23:
Nothing to backup, tear it down! 👿

vedran-dobos, 2016-08-27:
Done.

closed 2016-08-27 by vedran-dobos